kriswiner/LSM9DS0 #7: sampling frecuency

mutenroch, 2015-01-22:
First of all, thank for your work. Really inspiring

Please excuse if it is so basic what i'm asking:

I read in your papers that the filtering rate is well over 100hz for every controller. I uploaded your code to an arduino uno and i get data updated far below that fecuency (serial readings also) , could say it refresh more or less 2-3hz¡
I'm just trying to stabilize the roll angle of a device and it is so so so slow, 2 or 3 movements per second. (maybe i could avoid any other calculations, don't know exactly where but could try and error for ever if needed)
I must be missing something.
if you think this question is too basic to be attended i'll understand and continue reading about this theme, but any help would be really appreciated.
Thanks in advance

kriswiner, 2015-01-23:
You are using and Arduino Uno with an LSM9DS0? Which board are you using?
Are you using the interrupts for data ready detection or polling the status
registers?

I hope you have a 5 V tolerant sensor board as this sensor is 3V6 V or less.

One problem might be your integer types. The Teensy is a 32-bit processor
and the Arduino is 8 bit. So some subtle changes in the integer type might
change the speed.

Are you using my code for the LSM9DS0? Which one. One thing that might help
is to run the I2C at 400 kHz instead of 100 kHz. You get about a factor of 4
speed up. Without knowing your setup or code details its hard to diagnose
the problem any further.

Kris

<https://github.com/kriswiner/LSM9DS0/issues/7> .
<https://github.com/notifications/beacon/AGY1qpsrQnSiu0pnSM0ZtoPE5uga9bCbks5
nkVH2gaJpZM4DV_1x.gif>

mutenroch, 2015-01-23:
thanks Kris for quick answer.

The hardware is Arduino 1 with LSM9DS0 using a logic converter 3v-5v.
The firs code i used was the simplest one from sparkfun which uses just acc. data it was very quick but jittery.
Then i begun searching for filtered codes like yours. I used your LSM9DS0_AHRS.ino with your
SFE_LSM9DS0.h (both the latest-8month) as the come...
perhaps i have bad conections in the breackout board or something that doesn't trigger the interrupts properly....will check (multimeter: intr-ground??).

would like to know what part of the code can i delete/comment to improve speed due that i just need roll leveling also...I did with those lines about lcd already.

thanks again¡

kriswiner, 2015-01-23:
Even with a 16 MHz Arduino you should be able to get fusion rates well above 100 Hz. My recommendation is to start with a stripped down code and check how fast you can just get the data out. You can start with this code since it is easier to modify:

https://github.com/kriswiner/LSM9DS0/tree/master/Teensy3.1/LSM9DS0-MS5637

You will have to modify the wire.xxxx calls and you can eliminate all the MS5637 stuff.

If you start with just the basics you can add more and see how the speed is affected.

mutenroch, 2015-05-10:
Hi Kris,

followed your advice and began from striped version so I could isolate the problem.

The thing is that changing serial print/display rate I solved the problem.
Don't really understand why, as long as I know, serial rate is for comunicating with a pc (i.e) and data rate is usable for moving, in my case, a servo, but that is what i did and works¡

// Serial print and/or display at 0.5 s rate independent of data rates
    delt_t = millis() - count;
    if (delt_t > 10)  (changed delt_t>xxx)

kriswiner, 2015-05-10:
That's pretty odd but I have seen this kind of thing before. It has to do with the integer type of the variables delt_t and count. There is a difference between AVR and ARM machines at work here but i don't claim to understand it.