kriswiner/LSM9DS0 #12: Different question about serious drift

mmarkell, 2017-04-10:
Hello!

First of all, thank you for the code you've written for the LSM9DS0. It's taught me a lot and I am very grateful! 

However, sadly, something is going very wrong with my LSM9DS0 measurement drift.

Basically, let's say I hold the device at rest. The yaw, pitch, and roll start at some value (arbitrary seeming, definitely not zeroed for a flat and upward facing orientation, for instance.)

Next, they drift, slowly at first, and then in very large jumps. Within a few minutes, the yaw pitch and roll I'm calculating might go in complete circles, even though the device is stationary.

It's not totally inaccurate - it measures acceleration quite well from point A to point B, and knows in general which way I am turning the device. However, the drift over time is pretty insane. 

Steps I have taken: I am using a mahony filter, and recently implemented the calLSM9DS0 method from your code.

I would appreciate any advice you may have - do you think my device is faulty? 

Thank you for your help!

kriswiner, 2017-04-10:
Likely you need to caliubrate the magnetometer.

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

mmarkell, 2017-04-10:
> Effective-Magnetometer-Calibration

Taking a look now.

I was looking at
https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_BMP280_BasicAHRS_t3.ino
and I noticed that you have
float magCalibration[3] = {0, 0, 0}; // Factory mag calibration and mag bias
Then, you use this as a multiplicative factor in numerous equations.
However, you never set it to anything other than 0,0,0. Then, wouldn't any
operation being multiplied simply equal 0? Maybe I'm missing something

[image: Brown logo] <http://brown.edu/>
*Michael Markell ‘18*
B.S. Computer Science Candidate
302 229 0419

kriswiner, 2017-04-10:
> BMP280_BasicAHRS_t3.ino
> 302 229 0419 <(302)%20229-0419>

Yeah, your missing the mag calibration functions. One is in initAK8963 and
one is in magcalibration.  But try this instead anyway:

https://github.com/kriswiner/LSM9DS0/blob/master/Teensy3.1/LSM9DS0-MS5637/LSM9DS0_MS5637_Mini_Add_On.ino

closed 2024-11-02 by mmarkell