kriswiner/LSM9DS0 #13: Human movement

vitorsotero, 2017-09-10:
Hi,

I'm actually development a system that can evaluate human movements. I have a LSM9DS0 and a MPU6050 in hands.

I need to know not only wich movement was made, but the direction too. For this reason I think the LSM9DS0 is better (magnetometer). 

I was reading about quaternion and rotation matrix, but it's a very complex subject for me. I just need the acceleration (x, y and z) calibrated. Any suggestion?

Thanks,

Vitor Sotero

kriswiner, 2017-09-10:
If you just need acceleration then why the LSM9DS0?

If you need an absolute orientation estimation, then this is what you want:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

vitorsotero, 2017-09-10:
I need the velocity (I am integrating The acceleration). But the data
fronteira a 9DOF sensor is more accurate... or not?

My big question is: How to use the fusion data to get the "calibrated
acceleration"?

https://www.tindie.com/products/onehorse/ultimate-
sensor-fusion-solution/?pt=full_prod_search

> Hi,
>
> I'm actually development a system that can evaluate human movements. I
> have a LSM9DS0 and a MPU6050 in hands.
>
> I need to know not only wich movement was made, but the direction too. For
> this [...]
AGY1qo66Pe8lseh8iC3dPDOmDG4HSlijks5shBcRgaJpZM4PSYaA>

—
or mute
the thread

kriswiner, 2017-09-10:
What do you mean by calibrated acceleration? Do you mean linear
acceleration, with gravity removed?

You need the quaternions for that, but it is straightforward.

vitorsotero, 2017-09-10:
> Ww59FnRHrphIfQnNHFyieWL8CYxks5shBpugaJpZM4PSYaA

Linear acceleration and the direction

vitorsotero, 2017-10-01:
Hi Kris!

From now I'm getting the quaternion from the LSM9DS0. I calibrated the magnetometer removing hard and soft Iron effects. 

From now, I need to rotate my acceleration data (x, y and z). Do you think it's better to do with a rotation matrix (from quaternion) or the quaternion itself (q*accelData*q-¹)? Do you have any previous code?

My code to calibrate the mag:

void MagCalib()
{
   while (calibrated != true)//calibrate magnetometer http://www.camelsoftware.com/2016/03/13/imu-maths-calculate-orientation-pt3/
  {
    for (int i=0;i<10000;i++)
    {
    dof.readMag();
    min_x=min(min_x,dof.mx); min_y=min(min_y,dof.my); min_z=min(min_z,dof.mz);
    max_x=max(max_x,dof.mx); max_y=max(max_y,dof.my); max_z=max(max_z,dof.mz);
//    Serial.println(i);
//    Serial.print(min_x);Serial.print(", ");Serial.print(min_y);Serial.print(", ");Serial.println(min_z);
//    Serial.print(max_x);Serial.print(", ");Serial.print(max_y);Serial.print(", ");Serial.println(max_z);
    }
        
    mag_bias_x=(max_x + min_x)/2;//for hard iron error
    mag_bias_y=(max_y + min_y)/2;
    mag_bias_z=(max_z + min_z)/2;
    

    max_scale_x = max_x - mag_bias_x; min_scale_x = min_x - mag_bias_x;
    max_scale_y = max_y - mag_bias_y; min_scale_y = min_y - mag_bias_y;
    max_scale_z = max_z - mag_bias_z; min_scale_z = min_z - mag_bias_z;

    avg_x = max_scale_x + (min_scale_x*-1); avg_x = avg_x/2;
    avg_y = max_scale_y + (min_scale_y*-1); avg_y = avg_y/2;
    avg_z = max_scale_z + (min_scale_z*-1); avg_z = avg_z/2;
    avg_total = avg_x + avg_y + avg_z; avg_total = avg_total/3;

    scale_x = avg_total/avg_x; scale_y = avg_total/avg_y; scale_z = avg_total/avg_z;
        
    //Serial.print("Hard error(x,y,z): ");Serial.print(mag_bias_x);Serial.print(", ");Serial.print(mag_bias_y);Serial.print(", ");Serial.println(mag_bias_z);
    //Serial.print("Soft error(x,y,z): ");Serial.print(scale_x);Serial.print(", ");Serial.print(scale_y);Serial.print(", ");Serial.println(scale_z);
    calibrated = true;
    
    }

 if (digitalRead(INT2XM)) { // When new magnetometer data is ready
    dof.readMag();           // Read raw magnetometer data
//    mx = dof.calcMag(dof.mx);      // Convert to Gauss and correct for calibration
//    my = dof.calcMag(dof.my);
//    mz = dof.calcMag(dof.mz);
    mx = dof.mx;    
    my = dof.my;
    mz = dof.mz;
    mx -= mag_bias_x; //hard iron bias
    my -= mag_bias_y; 
    mz -= mag_bias_z;
    mx *= scale_x; //soft iron bias 
    my *= scale_y;
    mz *= scale_z;
}

Thanks!!

kriswiner, 2017-10-01:
> calculate-orientation-pt3/

Not sure I understand what you mean by "rotate my acceleration data".

vitorsotero, 2017-10-01:
I mean "rotate" in function to the Earth's reference, and not in function
to the sensor reference. To correct the accelerometer data when the person
makes a "randon" movement with the arm.

kriswiner, 2017-10-01:
Usually, with a 9 DoF sensor, the quaternions output from a proper fusion
filter are absolute with respect to True North and Gravity. So I would
derive anu accelerations you need from the quaternions.

vitorsotero, 2017-10-19:
Hi, Kris!

I am now using quaternion to generate a rotation matrix. And then, I rotate the Accelerometer data to true North.

But that's a issue that I I'm facing. On the figure, there are 3 axes: xAccel_rotated (blue), yAccel_rotated (red) and zAccel_rotated (green). And two circles: the red was when I tilted the sensor on the W/E global system. The black when I tilted on the N/S global system.

Do you have an idea why this occurs (black circle)?? 

Here some code about my rotation matrix:
[image]

`rotM[0] = 2.0f*(q[0]*q[0]) - 1 + 2.0f*(q[1]*q[1]);
rotM[1] = 2.0f*(q[1]*q[2]) - 2.0f*(q[3]*q[0]);
rotM[2] = 2.0f*(q[1]*q[3]) + 2.0f*(q[2]*q[0]);
rotM[3] = 2.0f*(q[1]*q[2]) + 2.0f*(q[3]*q[0]);
rotM[4] = 2.0f*(q[0]*q[0]) - 1 + 2.0f*(q[2]*q[2]);
rotM[5] = 2.0f*(q[2]*q[3]) - 2.0f*(q[1]*q[0]);
rotM[6] = 2.0f*(q[1]*q[3]) - 2.0f*(q[2]*q[0]);
rotM[7] = 2.0f*(q[2]*q[3]) + 2.0f*(q[1]*q[0]);
rotM[8] = 2.0f*(q[0]*q[0]) - 1 + 2.0f*(q[3]*q[3]);
accel_R[1] = rotM[0]*ax + rotM[1]*ay + rotM[2]*az;
accel_R[2] = rotM[3]*ax + rotM[4]*ay + rotM[5]*az;
accel_R[3] = rotM[6]*ax + rotM[7]*ay + rotM[8]*az;`

Thanks again,

Vitor Sotero

kriswiner, 2017-10-19:
Hi Vitor,

I don't understand what you are doing...

vitorsotero, 2017-10-19:
Summing up:

- I will evaluate human movement (using accelerometer data).
- I'm rotating the original accelerometer data using a rotation matrix from a quaternion.

kriswiner, 2017-10-19:
I guess I will ask you the usual questions:

How did you calibrate your sensors?

Are you feeding the Madgwick function the data in the proper order?

I suspect even if the answers are yes, then somehow your rotation matrices
are not quite right. But I am not sure i can check them for you...

closed 2017-10-19 by vitorsotero