kriswiner/LSM9DS0 #14: About Orientation

Bobur-kobilov, 2017-11-27:
Hi Kris,
Thank you a lot for the source code!

I implemented Madgwick source code with LSM9DS0 and the data that I get are really good but I have a problem with aligning the sensor orientation with global orientation. For instance, for visualization, I sent quaternion data to unity 3d and when I move the sensor in pitch direction, in the unity character is moving in yaw direction. Please could you help how to feed quaternion values to 3d character

kriswiner, 2017-11-27:
Well, you need to figure out which direction is North in unity 3d and then
choose the sensor order accordingly into the Madwick filter.

Bobur-kobilov, 2017-11-29:
Hi Kris,
Thank you for you reply! 
I have been wondering how I can calibrate LSMD90 IMU magnetometer

kriswiner, 2017-11-29:
https://github.com/kriswiner/LSM9DS0/blob/master/Teensy3.1/MiniShield/LSM9DS0_MS5611_BasicAHRS_t3.ino

Bobur-kobilov, 2017-11-30:
Thank you kris! I implemented and calibration is working well!
I have a question about beta value! If you can help me figure out, it would be appreciated!
1. What is the function of beta value? 
2. For each sensor, do I have to tweak beta value to eliminate drift?