kriswiner/LSM9DS0 #15: Yaw angle calculation about z-axis.

paresh3108, 2019-11-11:
Hi Kris
    Personally i really found this code very useful for calibration and consequently to obtain the values.
However, I need your help in finding yaw angle about z-axis. In your code, it seems to be specified about x-axis. Can you please help me out?
    Also do you have any good links to understand the relation between Eulerian and Quaternions angles.

kriswiner, 2019-11-11:
Google "euler angles from quaternions", lots of links there.

The Yaw is typically defined as rotation about the z-axis so I don't quite
understand what you are asking for.

paresh3108, 2019-11-11:
Thanks Kris for your reply,
But I mentioned it like that because in your code you said that:
Yaw is the angle between Sensor x-axis and Earth magnetic North 
and I want you to find angle about z axis, I hope you are getting my problem now.

kriswiner, 2019-11-11:
Yaw is angle made *between* x-axis and true North, changes in yaw so
defines are rotations *about* the z-axis. So I still do not understand your
question.

paresh3108, 2019-11-11:
Thanks, No Problem 
Anyways I got my solution after some edits